PRBonn/LiDAR-MOS #1: TF/Tensorboard version mismatch

guragamb, 2021-07-16:
Hi there, thanks for releasing the code, the paper was a very interesting read! 

I'm getting errors when trying to run MOS with SalsaNext related to the TF/Tensorboard versions. As per [the config file](https://github.com/PRBonn/LiDAR-MOS/blob/main/mos_SalsaNext/salsanext_cuda10.yml), TF and Tensorboard versions 1.13.1 but in LiDAR-MOS/mos_SalsaNext/train/common/logger.py line 18, you use `self.writer = tf.summary.create_file_writer(log_dir)` which seems to be the way to call Tensorboard using TF 2.0 (https://github.com/mingyuliutw/UNIT/issues/56#issuecomment-589504964, https://www.tensorflow.org/tensorboard/migrate). 

This is the exact error I get: 
```
No pretrained directory found.
Copying files to /host-machine/ml/LiDAR-MOS/logs/logs/2021-7-16-18:10salsanext for further reference.
Sequences folder exists! Using sequences from /host-machine/semKITTI/Data/SemanticKitti/dataset/sequences
parsing seq 00
parsing seq 01
parsing seq 02
parsing seq 03
parsing seq 04
parsing seq 05
parsing seq 06
parsing seq 07
parsing seq 09
parsing seq 10
Using 19130 scans from sequences [0, 1, 2, 3, 4, 5, 6, 7, 9, 10]
Sequences folder exists! Using sequences from /host-machine/semKITTI/Data/SemanticKitti/dataset/sequences
parsing seq 08
Using 4071 scans from sequences [8]
Loss weights from content:  tensor([  0.0000,   1.0210, 296.4371])
Depth of backbone input =  6
Traceback (most recent call last):
  File "./train.py", line 177, in <module>
    trainer = Trainer(ARCH, DATA, FLAGS.dataset, FLAGS.log, FLAGS.pretrained,FLAGS.uncertainty)
  File "../../tasks/semantic/modules/trainer.py", line 130, in __init__
    self.tb_logger = Logger(self.log + "/tb")
  File "../../common/logger.py", line 18, in __init__
    self.writer = tf.summary.create_file_writer(log_dir)
AttributeError: module 'tensorflow._api.v1.summary' has no attribute 'create_file_writer'
```

Chen-Xieyuanli, 2021-07-17:
Yes, it's a version conflict of TF/Tensorboard. You may have a look at the Tensorboard summary writer error here https://github.com/mingyuliutw/UNIT/issues/56

Chen-Xieyuanli, 2021-08-14:
Hey @guragamb, did you resolve the version conflict problem?

Chen-Xieyuanli, 2021-09-02:
Since there is no update. I would like to close this issue.

closed 2021-09-02 by Chen-Xieyuanli

e1339g, 2022-09-09:
@Chen-Xieyuanli Thank you for sharing your code. Unfortunately, I have a similar error. I've changed it FileWriter and then received another error which says "FileWriter has no attribute as_default" and the scalar summary function does not work. 
@guragamb Please help me to solve this. 
Finally, is this written in tensorflow 1 oer 2? Thank you.