PRBonn/LiDAR-MOS #3: 方法的适用性

zgq6667, 2021-07-20:
您好，我是中国的一位研究生，我想问一下，您的方法可以用在其他SLAM框架，如loam或者lego loam

Chen-Xieyuanli, 2021-07-21:
@zgq6667, 你好！感谢关注我们的代码！我们的算法使用CNN直接对点云输入进行处理，可以移除当前点云中的动态物体，因此理论上所有可以在KITTI Odometry上运行的激光雷达SLAM都能使用我们算法处理之后的点云，并可能得到SLAM性能上的提升。

Q: The generalization of the proposed method. Can it be used with other SLAM methods, like LOAM or LeGO-LOAM?
A: Yes, our method can directly segment the moving objects from the LiDAR scan with CNN. Theoretically, all LiDAR-based SLAM that can be run with the KITTI-Odometry data can use the post-processed LiDAR scans cleaned by our method and may improve the odometry/SLAM results.

daijiaoshou, 2021-07-26:
您好，想咨询下目前算法在处理后输出的是什么类型的点云数据? ros下一般是sensor::msgs PointCloud2

Chen-Xieyuanli, 2021-07-27:
@daijiaoshou 在我们目前提供的demo中，被清理的点云以KITTI点云数据.bin格式存储，具体存储代码在这https://github.com/PRBonn/LiDAR-MOS/blob/52a6fca4a6b71921fa17623b1902528e7a84a1ea/utils/scan_cleaner.py#L81
你可以通过修改这行代码存储成你想要的格式。

Q: What is the format of the post-processed point cloud using MOS?
A: In the current demo, we save the post-processed point cloud in the binary format following the KITTI dataset. One could modify the line of code for saving [here](https://github.com/PRBonn/LiDAR-MOS/blob/52a6fca4a6b71921fa17623b1902528e7a84a1ea/utils/scan_cleaner.py#L81) and easily save it in the desired format.

hanbing123-cool, 2021-08-13:
请问您实现在slam上的应用了吗？

Chen-Xieyuanli, 2021-08-13:
Hey @hanbing123-cool, as shown in the [applications](https://github.com/PRBonn/LiDAR-MOS#applications), we tested the cleaned point clouds with SuMa and got much better SLAM results.

Chen-Xieyuanli, 2021-09-28:
Since there is no further update, I would like to close this issue. If you have any further questions, please feel free to ask me to reopen it.

closed 2021-09-28 by Chen-Xieyuanli

zyx-99, 2022-12-02:
> 您好，我是中国的一位研究生，我想问一下，您的方法可以用在其他SLAM框架，如loam或者lego loam
您好，请问您成功用这个跑出的数据运行ALOAM LIOSAM这些SLAM算法了嘛？

zyx-99, 2023-08-24:
> 请问您实现在slam上的应用了吗？

请问您成功实现了嘛？