PRBonn/LiDAR-MOS #4: Inferring the test dataset using SalsaNext pretrain model

songw-zju, 2021-07-25:
Hi, thanks for your great work! 

I'm getting errors when trying to infer the test dataset with SalsaNext pretrain model. It runs fluently on train and valid dataset. But when I run it on the test dataset, I can only get the prediction result of sequence 11, and then I get the following error
```
Traceback (most recent call last):
  File "infer.py", line 145, in <module>
    user.infer()
  File "../../tasks/semantic/modules/user.py", line 113, in infer
    to_orig_fn=self.parser.to_original, cnn=cnn, knn=knn)
  File "../../tasks/semantic/modules/user.py", line 134, in infer_subset
    for i, (proj_in, proj_mask, _, _, path_seq, path_name, p_x, p_y, proj_range, unproj_range, _, _, _, _, npoints) in enumerate(loader):
  File "/home/ws/anaconda3/envs/py37/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 345, in __next__
    data = self._next_data()
  File "/home/ws/anaconda3/envs/py37/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 838, in _next_data
    return self._process_data(data)
  File "/home/ws/anaconda3/envs/py37/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 881, in _process_data
    data.reraise()
  File "/home/ws/anaconda3/envs/py37/lib/python3.7/site-packages/torch/_utils.py", line 395, in reraise
    raise self.exc_type(msg)
IndexError: Caught IndexError in DataLoader worker process 1.
Original Traceback (most recent call last):
  File "/home/ws/anaconda3/envs/py37/lib/python3.7/site-packages/torch/utils/data/_utils/worker.py", line 178, in _worker_loop
    data = fetcher.fetch(index)
  File "/home/ws/anaconda3/envs/py37/lib/python3.7/site-packages/torch/utils/data/_utils/fetch.py", line 44, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/ws/anaconda3/envs/py37/lib/python3.7/site-packages/torch/utils/data/_utils/fetch.py", line 44, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "../..//tasks/semantic/dataset/kitti/parser.py", line 244, in __getitem__
    exec("residual_file_" + str(i+1) + " = " + "self.residual_files_" + str(i+1) + "[seq][index]")
  File "<string>", line 1, in <module>
IndexError: list index out of range
```

HenryJunW, 2021-11-16:
Same problem, and did you fix it? Thanks

HenryJunW, 2021-11-16:
Fixed the problem by modifying setting up the custom residual_image path since my generated residual images are not in the root directory.  
Just for the future reference, line https://github.com/PRBonn/LiDAR-MOS/blob/7ccbb63b4ee7c40195b35dd0dddd71473fae25b1/mos_SalsaNext/train/tasks/semantic/dataset/kitti/parser.py#L160.

Chen-Xieyuanli, 2021-11-16:
> Fixed the problem by setting up the custom residual_image path since my generated residual images are not in the self.root directory. Just for the future reference,
> 
> https://github.com/PRBonn/LiDAR-MOS/blob/7ccbb63b4ee7c40195b35dd0dddd71473fae25b1/mos_SalsaNext/train/tasks/semantic/dataset/kitti/parser.py#L160
> 
> .

Hey @HenryJunW, thanks for using our code and for your kindly report.

closed 2021-11-16 by Chen-Xieyuanli

skylineover, 2023-06-25:
What are the specific steps? Can you give me more details? Thank you