PRBonn/LiDAR-MOS #6: 新手提问 请问可以使用自定义数据集么 用来剔除当前数据集包含以外的物体？谢谢

aprilliuwei, 2021-07-28:


Chen-Xieyuanli, 2021-07-28:
Hey @aprilliuwei, 感谢给我们留言！我们目前正在进行一项开放世界集算法研究(检测训练集中没有的物体)，目前论文正在盲审过程中，如被录用，我们会第一时间放在主页，欢迎关注我们主页与我们保持联系。我们也可以私底下交流具体细节，我的邮箱是：xieyuanli.chen@igg.uni-bonn.de

Q: How to use it with a self-defined dataset and remove objects those are not in the training data?
A: We just submitted a work to detect and segment objects in an open-world set. The submission is still under double-blind reviewing and we will release the paper after the review on our website. To discuss more the open-world set segmentation, one could send emails to me xieyuanli.chen@igg.uni-bonn.de

Chen-Xieyuanli, 2021-08-14:
Hey @aprilliuwei, since this issue is not about MOS, I would like to close it. If you have any further questions, please feel free to send me msg.

closed 2021-08-14 by Chen-Xieyuanli