PRBonn/LiDAR-MOS #10: bash: ./train.sh: 没有那个文件或目录

jiaoluonanhai, 2021-07-29:
如题，我在尝试运行程序时提示没有文件，请问是忘了上传了吗

  $ cd mos_SalsaNext/train/tasks/semantic
  $ ./train.sh -d path/to/kitti/dataset -a salsanext_mos.yml -l path/to/log -c 0  # the number of used gpu cores

Chen-Xieyuanli, 2021-07-29:
Hey @jiaoluonanhai,  感谢使用我们的代码，`.\train.sh`文件在`mos_SalsaNext`根目录，我现在已经修改了readme.md。你可以再尝试训练一下。

jiaoluonanhai, 2021-07-29:
`root@user:~/LiDAR-MOS-main/mos_SalsaNext# ./train.sh -d path/to/kitti/dataset -a salsanext_mos.yml -l path/to/log -c 0  # the number of used gpu cores
./train.sh: 4: cd: can't cd to path/to/kitti
./train.sh: 4: cd: can't cd to path/to
/home/user/.local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:526: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint8 = np.dtype([("qint8", np.int8, 1)])
/home/user/.local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:527: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
/home/user/.local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:528: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint16 = np.dtype([("qint16", np.int16, 1)])
/home/user/.local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:529: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
/home/user/.local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:530: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint32 = np.dtype([("qint32", np.int32, 1)])
/home/user/.local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:535: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  np_resource = np.dtype([("resource", np.ubyte, 1)])
Opening arch config file /home/user/LiDAR-MOS-main/mos_SalsaNext/salsanext_mos.yml
Opening data config file config/labels/semantic-kitti-mos.yaml
Depth of backbone input =  6
----------
INTERFACE:
dataset /dataset
arch_cfg /home/user/LiDAR-MOS-main/mos_SalsaNext/salsanext_mos.yml
data_cfg config/labels/semantic-kitti-mos.yaml
uncertainty False
Total of Trainable Parameters: 6.71M
log /log/logs/2021-7-29-20:17
pretrained 
----------

----------

No pretrained directory found.
Copying files to /log/logs/2021-7-29-20:17 for further reference.
Traceback (most recent call last):
  File "./train.py", line 177, in <module>
    trainer = Trainer(ARCH, DATA, FLAGS.dataset, FLAGS.log, FLAGS.pretrained,FLAGS.uncertainty)
  File "../../tasks/semantic/modules/trainer.py", line 108, in __init__
    shuffle_train=True)
  File "../..//tasks/semantic/dataset/kitti/parser.py", line 432, in __init__
    gt=self.gt)
  File "../..//tasks/semantic/dataset/kitti/parser.py", line 124, in __init__
    raise ValueError("Sequences folder doesn't exist! Exiting...")
ValueError: Sequences folder doesn't exist! Exiting...
`
这是我在mos_SalsaNext目录下运行的结果请问是因为我缺少文件吗 还是路径问题？ @Chen-Xieyuanli

Chen-Xieyuanli, 2021-07-29:
你要根据自己数据放置的情况设置好path/to/kitti/dataset（数据集路径）和path/to/log（训练结果存储路径）

jiaoluonanhai, 2021-07-29:
感谢哥，原来那个问题解决了 请问老哥咱这个程序使用的是哪个版本的tensorflow;我这边提示'AttributeError: module 'tensorflow._api.v1.summary' has no attribute 'create_file_writer''

Chen-Xieyuanli, 2021-07-29:
> 感谢哥，原来那个问题解决了 请问老哥咱这个程序使用的是哪个版本的tensorflow;我这边提示'AttributeError: module 'tensorflow._api.v1.summary' has no attribute 'create_file_writer''

之前的问题解决了就好！这确实是tensor flow版本问题，我用的tensorflow2.0，你可以在这里找到解决方案https://github.com/mingyuliutw/UNIT/issues/56 
由于这个issue之前有人提过了#1，我把这个issue关掉了，我们可以在#1中继续讨论。
如果觉得我的回答有帮助，请麻烦帮我们点个赞;-)

closed 2021-07-29 by Chen-Xieyuanli

jsjcn, 2023-10-13:
train.sh: 行 14: --output_dir：未找到命令，这咋办啊