PRBonn/LiDAR-MOS #8: 环境配置

whuzs, 2021-07-29:
您好,我想问一下LiDAR-MOS训练和推理的环境配置要求?谢谢

Chen-Xieyuanli, 2021-07-29:
Hey @whuzs, 感谢使用我们的代码！我们的代码是基于已有的语义分割网络构建的，如果使用RangeNet++你可以在[mos_RangeNet]((https://github.com/PRBonn/LiDAR-MOS/tree/main/mos_RangeNet))中找到相关配置，如果使用SalsaNext你可以在[mos_SalsaNext](https://github.com/PRBonn/LiDAR-MOS/tree/main/mos_SalsaNext)中找到相关配置。

Q: Questions about the setup of LiDAR-MOS training and inferring
A: We built our method atop existing semantic segmentation networks. To use RangeNet++, you could find the setup in [mos_RangeNet]((https://github.com/PRBonn/LiDAR-MOS/tree/main/mos_RangeNet)). To use SalsaNext, you could find the setup in [mos_SalsaNext](https://github.com/PRBonn/LiDAR-MOS/tree/main/mos_SalsaNext).

jiaoluonanhai, 2021-07-30:
@Chen-Xieyuanli  大佬训练的时候使用的是什么显卡啊
我的是1650ti 提示内存不足，感觉训练是不是得需要个相对较好的GPU？
以下是错误提示
RuntimeError: CUDA out of memory. Tried to allocate 1010.00 MiB (GPU 0; 3.82 GiB total capacity; 1.41 GiB already allocated; 1019.75 MiB free; 1.51 GiB reserved in total by PyTorch)

Chen-Xieyuanli, 2021-07-30:
> @Chen-Xieyuanli 大佬训练的时候使用的是什么显卡啊
> 我的是1650ti 提示内存不足，感觉训练是不是得需要个相对较好的GPU？
> 以下是错误提示
> RuntimeError: CUDA out of memory. Tried to allocate 1010.00 MiB (GPU 0; 3.82 GiB total capacity; 1.41 GiB already allocated; 1019.75 MiB free; 1.51 GiB reserved in total by PyTorch)

我测试过1080ti, 2080ti, Quadro 4000, Quadro 6000和Titan，其中1080ti，2080ti和Titan每块可以跑两个batch，其他的每块只能一个batch，你可以根据自己的显卡调整batchsize大小

Chen-Xieyuanli, 2021-08-14:
Since there is no more update about this issue, I would like to close it. If you have any questions, please send me msgs.

closed 2021-08-14 by Chen-Xieyuanli