PRBonn/LiDAR-MOS #11: labels

May-sysu, 2021-07-31:
can we predict the moving object without lables?

Chen-Xieyuanli, 2021-07-31:
Hey @May-sysu, thanks for following our work. Our method currently using the proposed binary labels to achieve MOS.
If you want to remove dynamics without using labels, you may have a look at the map-based approaches, which can also detect moving objects without training.

May-sysu, 2021-08-02:
yep，i know that。however, supervised classification is out-performer.  can we generate the moving object predictions with pretrained model in different scene？for example, we use pretrained model based on kitti datasets to predict the moving object in the mining area, like transfer application. and what do we need to focus on ? thanks for your answer

Chen-Xieyuanli, 2021-08-02:
Good idea! We didn't test it so far. You may give it a try. The only problem I see here is that if the types of LIDAR sensors change, it usually cannot generalize very well.

Chen-Xieyuanli, 2021-08-14:
Hey @May-sysu, thanks for the suggestion again! Since this is more about future work, I therefore would like to close this issue. If you are interested in this topic, we could discuss it further via email.

closed 2021-08-14 by Chen-Xieyuanli

May-sysu, 2021-08-14:
thanks for your generosity。after converting kitti-64 to kitti-16, this algorithm still works well。however, in the mining area, there is no accurate pose information ,which is a great blow to it because of residual image. whatever，i think this light algorithm maybe is better in the object detection.

Chen-Xieyuanli, 2021-08-14:
Great to know that it works with different types of LIDAR sensors. We could also discuss this more while WeChat. My WeChat number is : Chen_Xieyuanli 

I will appreciate it if you could also star this repo if you find it useful. Thanks!