PRBonn/LiDAR-MOS #14: Can the program remove frames of dynamic objects?If so, where?

hanbing123-cool, 2021-08-13:
Can the program remove frames of dynamic objects?If so, where?

Chen-Xieyuanli, 2021-08-13:
Hey @hanbing123-cool, I didn't get this question. If you mean to remove dynamics objects in LIDAR frames, the answer is yes.
As showing in the demo and applications, we provide the script to clean LIDAR frames here: https://github.com/PRBonn/LiDAR-MOS#clean-the-scans

hanbing123-cool, 2021-08-14:
After I have executed "python3 utils/scan_cleaner.py", why do I still have dynamic objects in the frame under "clean_SCANS" when playing with your visualizer?

hanbing123-cool, 2021-08-14:
How can I get a frame that removes a dynamic object?And then send him to the SLAM system

Chen-Xieyuanli, 2021-08-14:
> After I have executed "python3 utils/scan_cleaner.py", why do I still have dynamic objects in the frame under "clean_SCANS" when playing with your visualizer?

Could you please show the results you got?

hanbing123-cool, 2021-08-14:
I used the command "python3 utils/visualize_mos.py -d data" +"clean_scans" (including removing dynamic frames) to play back,I thought "python3 utils/scan_cleaner.py" generated the frames with the removal of the motion. I now want to know what folder the frames with the removal of the motion are in.Sorry I haven't looked at the source code yet.

Chen-Xieyuanli, 2021-08-16:
The visualizer we provide so far can only visualize the move object segmentation results, which only show scans frame by frame. You could use scan_cleaner to clean the scan but it then cannot be visualized via visualize_mos, since the number of points is different from the raw point cloud.

We didn't release the visualizer for the clean map so far and we may release it later.

hanbing123-cool, 2021-08-23:
Can you recommend a data set that has label files and IMU and GNSS data? Because the data set you provided does not have IMU information, the data set I downloaded before does not have label file. My SLAM can't be tested

hanbing123-cool, 2021-08-23:
If I use my own data set, I do not have a label file, can I use your program to remove dynamic objects? Use your program to speculate whether there is a label file? Looking forward to your reply.

Chen-Xieyuanli, 2021-08-23:
> Can you recommend a data set that has label files and IMU and GNSS data? Because the data set you provided does not have IMU information, the data set I downloaded before does not have label file. My SLAM can't be tested

We will probably release our dataset this year. It will include all the data.

> If I use my own data set, I do not have a label file, can I use your program to remove dynamic objects? Use your program to speculate whether there is a label file? Looking forward to your reply.

I don't how different is your data from the KITTI dataset. I therefore can not tell whether it works directly with the pre-trained model or not. You could anyway give it a try and I will appreciate it if you could later also let me know whether it works or not.
Another solution is that we also provide the point cloud labeling tool [here](https://github.com/jbehley/point_labeler). You could also some of your data and fine-tune our model to make it work in your specific environment.

Chen-Xieyuanli, 2021-09-28:
Since there is no further update, I would like to close this issue. If you have any further questions, please feel free to ask me to reopen it.

closed 2021-09-28 by Chen-Xieyuanli

zyx-99, 2022-12-02:
> If I use my own data set, I do not have a label file, can I use your program to remove dynamic objects? Use your program to speculate whether there is a label file? Looking forward to your reply.

您好，请问您成功用这个程序的输出跑通SLAM程序了嘛？可否交流一下呢