PRBonn/LiDAR-MOS #15: 关于多帧residual的问题

wangx1996, 2021-08-15:
您好，关于此工作，对于gen_residual_images.py中

      last_pose = poses[frame_idx - num_last_n]
      last_scan = load_vertex(scan_paths[frame_idx - num_last_n])
      last_scan_transformed = np.linalg.inv(current_pose).dot(last_pose).dot(last_scan.T).T
      last_range_transformed = range_projection(last_scan_transformed.astype(np.float32),
                                                range_image_params['height'], range_image_params['width'],
                                                range_image_params['fov_up'], range_image_params['fov_down'],
                                                range_image_params['max_range'], range_image_params['min_range'])[:, :, 3]
      
对于不同的num_last_n数量也只是取了对于当前帧num_last_n前的那一帧数据吗？而不是连续数据叠加？

wangx1996, 2021-08-15:
我知道了，多运行几遍gen_residual_images是吧

Chen-Xieyuanli, 2021-08-15:
@wangx1996 你好，感谢使用我们的代码。是的，提供的代码只能一次生成某一过去帧和现在帧的residual image。我们目前没有计划提供一次性生成所有residual image的代码，如果你感兴趣，欢迎提PR。

Q: Does the current script only generate the residual images between the last n-th frame and the current frame but not all the ones from in between?
A: Yes. One could generate the residual images in between by changing the parameters and running the script multiple times.

I think the answer is clear and therefore would like to close this issue. If you have any other questions, please feel free to leave me a message.

closed 2021-08-15 by Chen-Xieyuanli