PRBonn/LiDAR-MOS #16: 测试程序运行不起来

yst1, 2021-08-20:
[image]
当我运行提供的测试集时就一直停在这个页面，请问是什么问题呢？有人知道吗

yst1, 2021-08-20:
用的是文章提供训练好的模型和测试数据。

Chen-Xieyuanli, 2021-08-20:
Hey @yst1, 感谢使用我们的代码，你好像在用RangeNet的模型，我一会测试一下看能不能复现你的问题，你也可以用SalsaNext模型试一下，看会不会有同样问题。

Q: The evaluation paused with RangeNet backbone on the validation set.
A: Hey @yst1, thanks for using our code!
It seems you are using the RangeNet as the backbone. Could you please try the SalsaNext model and see whether you still have such a problem? I will give a test with the RangeNet backbone.

yst1, 2021-08-20:
你好，请问你训练SalsaNext模型的pytorch版本是多少？我这边用1.4.0的加载模型说版本太旧了。

Chen-Xieyuanli, 2021-08-20:
我用的1.5.1+cu101

closed 2021-08-24 by yst1