PRBonn/LiDAR-MOS #17: How can i train 'SalsaNext' successfully?（训练'SalsaNext'时侯出现了问题）

Sunghooon, 2021-08-23:
Hi, thanks for sharing your great code.
I'm just trying to do whole process of your works.
but I can't train SalsaNext, 

I tried : 
```
./train.sh -d ../../../../dataset/KITTI_dataset/velodyne_laser/dataset/ -a salsanext_mos.yml -l logs/ -c 0
```
learning process arrived at :
```
Lr: 5.944e-03 | Update: 2.381e-04 mean,3.611e-04 std | Epoch: [0][11370/19130] | Time 0.203 (0.204) | Data 0.030 (0.031) | Loss 0.3839 (0.2800) | acc 0.962 (0.980) | IoU 0.685 (0.517) | [7 days, 18:10:40]
```
and the error msgs i got :
```
proj_full = torch.cat([proj_full, torch.unsqueeze(eval("proj_residuals_" + str(i+1)), 0)])
RuntimeError: invalid argument 0: Sizes of tensors must match except in dimension 0. Got 2048 and 2058 in dimension 2 at ../aten/src/TH/generic/THTensor.cpp:711
```

Is the conda environment setting wrong? currently using :
```
tensorboard               1.13.1                   pypi_0    pypi
tensorboard-data-server   0.6.1                    pypi_0    pypi
tensorboard-plugin-wit    1.8.0                    pypi_0    pypi
tensorflow                1.13.1                   pypi_0    pypi
tensorflow-estimator      1.13.0                   pypi_0    pypi
```
Or, can you check again links `Collection of downloads` ? seems like can't access now.
Thanks you !

Chen-Xieyuanli, 2021-08-23:
Hey @Sunghooon, thanks for using our code.

It seems the residual images are incomplete.

For the collection of downloads, could you please try again by using the download link with another browser? It works for me with chrome. I will also try it outside our network to have another check.

hanbing123-cool, 2021-08-23:
Can you recommend a data set that has label files and IMU and GNSS data? Because the data set you provided does not have IMU information, the data set I downloaded before does not have label file. My SLAM can't be tested

Sunghooon, 2021-08-23:
Thanks for your quick response !
Let me double-check generating residual images process. thank you

Sunghooon, 2021-08-24:
Hello, @Chen-Xieyuanli, I solved the problem about training data.
(There was typo in my `data_preparing.yaml` file, thanks!)

and I briefly trained data, got `/predictions`, `/tb`, `arch_cfg/data_cfg.yaml`, `log.txt`, `SalsaNext` and `SalsaNext_train/valid_best`. But, when i tried `Inferring`, I got a error msg below :

```
Traceback (most recent call last):
  File "infer.py", line 144, in <module>
    user = User(ARCH, DATA, FLAGS.dataset, FLAGS.log, FLAGS.model,FLAGS.split,FLAGS.uncertainty,FLAGS.monte_carlo)
  File "../../tasks/semantic/modules/user.py", line 70, in __init__
    self.model.load_state_dict(w_dict['state_dict'], strict=True)
  File "/home/psx/anaconda3/envs/salsanext/lib/python3.7/site-packages/torch/nn/modules/module.py", line 777, in load_state_dict
    self.__class__.__name__, "\n\t".join(error_msgs)))
RuntimeError: Error(s) in loading state_dict for DataParallel:
	Missing key(s) in state_dict: "module.downCntx.conv1.weight", "module.downCntx.conv1.bias", "module.downCntx.conv2.weight", "module.downCntx.conv2.bias", "module.downCntx.bn1.weight", "module.downCntx

...
	Unexpected key(s) in state_dict: "downCntx.conv1.weight", "downCntx.conv1.bias", "downCntx.conv2.weight", "downCntx.conv2.bias", "downCntx.bn1.weight", "downCntx.bn1.bias", "downCntx.bn1.running_mean", "downCntx.bn1.running_var", "downCntx.bn1.num_batches_tracked", "downCntx.conv3.weight", "downCntx.conv3.bias", "downCntx.bn2.weight", "downCntx.bn2.bias",
...

```
 Did i skip some commands or settings? thanks for your reply !

Sunghooon, 2021-08-25:
I edited `user.py` and solved problem. (line70)
```
            # https://github.com/bearpaw/pytorch-classification/issues/27
            # comment by 'gftm'
            #self.model.load_state_dict(w_dict['state_dict'], strict=True)
            state_dict = w_dict['state_dict']
            from collections import OrderedDict
            new_state_dict = OrderedDict()

            for k, v in state_dict.items():
              if 'module' not in k:
                k = 'module.'+k
              else:
                k = k.replace('features.module.', 'module.features.')
              new_state_dict[k] = v
            self.model.load_state_dict(new_state_dict)
```

Chen-Xieyuanli, 2021-08-25:
Thank you @Sunghooon for reporting the solutions! 

The 'Unexpected key(s)' I think is actually caused by the incompatible Pytorch versions. I'm glad that you fixed it already.

It would be great for other users if you could also share your GPU setup and Pytorch version here.