PRBonn/LiDAR-MOS #18: Multiple input frames

Trainingzy, 2021-08-30:
Thank you for your great code!

I notice in `rangenet_mos.yaml` and `salsanext_mos.yaml` you only use one input frame. If I want to use multiple scans for training, how can I do it?

As far as I know, I need to change [n_input_scans](https://github.com/PRBonn/LiDAR-MOS/blob/69e85fc7f221cd856512d39500763973c40c580c/mos_RangeNet/tasks/semantic/rangenet_mos.yaml#L39) in both backbone and dataset, and [transform](https://github.com/PRBonn/LiDAR-MOS/blob/69e85fc7f221cd856512d39500763973c40c580c/mos_RangeNet/tasks/semantic/rangenet_mos.yaml#L105) in the dataset. What else do I need?

Trainingzy, 2021-08-30:
Btw, I am not sure about [this](https://github.com/PRBonn/LiDAR-MOS/blob/69e85fc7f221cd856512d39500763973c40c580c/mos_RangeNet/tasks/semantic/dataset/kitti/parser.py#L260), it seems that you use `proj_mask` of the last frame of a sequence to deal with all the scans?

Chen-Xieyuanli, 2021-08-30:
Hey @Trainingzy,  

Thanks for using our code!

> Thank you for your great code!
>
> I notice in `rangenet_mos.yaml` and `salsanext_mos.yaml` you only use one input frame. If I want to use multiple scans for training, how can I do it?
>
> As far as I [...]

Yes, those are exactly what you need to do. To directly use multiple scans, the most important changes is in the data loader and the size of the backbone.

> Another question is about the pose transform. When using a sequence of scans to train the model, why do you transform the pose to the last scan but not the first scan? [here](https://github.com/PRBonn/LiDAR-MOS/blob/69e85fc7f221cd856512d39500763973c40c580c/mos_RangeNet/tasks/semantic/dataset/kitti/parser.py#L180)

This is because we also want to integrate our method into the SLAM algorithm. There, the poses for the past scans are for free from the odometry, and for the current frame, only the past scans are available.

Chen-Xieyuanli, 2021-08-30:
> Btw, I am not sure about [this](https://github.com/PRBonn/LiDAR-MOS/blob/69e85fc7f221cd856512d39500763973c40c580c/mos_RangeNet/tasks/semantic/dataset/kitti/parser.py#L260), it seems that you use `proj_mask` of the last frame of a sequence to deal with all the scans?

Thanks for pointing this out. Yes, that could be problematic. I just use the mask to get rid of the invalid point in the current scan, without taking care of the invalid points of other scans. It could be improved later.

Trainingzy, 2021-08-30:
Thank you for your reply.

For the second question, it is exactly what confuses me. As u said, *For the current frame, only the past scans are available.*, if so, for a sequence of frames, I think we should change the pose of each frame to the first frame of this sequence, instead of the last frame as u did.

For the third question, I think we just need to first mask out invalid points of one scan and then concatenate it with the previous scans, like:
```
proj = proj * proj_mask.float()
proj_full = torch.cat([proj_full, proj])
```
I am not quite sure about it, what do u think?

Chen-Xieyuanli, 2021-08-30:
Good point! The argument why transfer all past scans to the current view point is that we care more about the MOS in the current observation. The solution is to estimate the pose for the current scan using ICP and then clean the scan and estimate the pose again. You could also project to the first scan. Once we have poses, the process is actually the same. If one could find a way to achieve MOS without using poses would be even better.

zxczrx123, 2021-08-31:
Thanks for your great work!
I also have a few questions about multiple input frames
1. Runtime. I am curious about that if the residual image method cloud be used online, especially when residual image nums N=8?(In your paper i saw it is 10ms for one residual image, does this mean that it takes 80ms for 8 residual images?) 
2. Residual generation. During training and testing, is it necessary to execute the **utils/gen_residual_images.py** N times for each seq? Is there a script to automate the process?

Trainingzy, 2021-09-06:
HI @Chen-Xieyuanli 
I found you change the raw pose data with pose 0 of each sequence [here](https://github.com/PRBonn/LiDAR-MOS/blob/69e85fc7f221cd856512d39500763973c40c580c/mos_RangeNet/tasks/semantic/dataset/kitti/parser.py#L140):
```
new_poses.append(T_velo_cam.dot(inv_frame0).dot(pose).dot(T_cam_velo))
```
I do not know why you multiply the pose with `inv_frame0`, and in 4D panoptic segmentation (also deal with sequence data of SemanticKITTI), they only do as `T_velo_cam.dot(pose).dot(T_cam_velo)` [here](https://github.com/MehmetAygun/4D-PLS/blob/b8963d4ac2bc65acbb42691cb1cec7c89a012f9d/datasets/SemanticKitti.py#L897).

Look forward to your reply~~

Chen-Xieyuanli, 2021-09-06:
> HI @Chen-Xieyuanli
> I found you change the raw pose data with pose 0 of each sequence [...]

Usually, the first pose of the SLAM algorithm would be an identity matrix. I dot pose0 inverse is in case that sometimes the first pose is not an identity matrix and may bring some issues.

Trainingzy, 2021-09-06:
Cool~~ Thanks for your patience and reply

closed 2021-09-06 by Trainingzy