PRBonn/LiDAR-MOS #19: 静动态效果不佳

yst1, 2021-08-31:
你好，我用你训练好的模型跑自己的16线雷达的数据，根据label输出对当前帧点云每个点区分静动态，最后输出静动态点云。但是效果是这样的，彩色的是动态点，请问可能是什么原因呢？
[image]

Chen-Xieyuanli, 2021-08-31:
@yst1 你好，感谢使用我们的代码。从提供的实验结果来看，预训练的模型在不同激光雷达上的泛化能力不行。可行的解决方案是对预训练进行微调再训练（fine tune）。可以你的数据集中标注少量动静态物体标签，然后再训练一下，结果应该能够得到提升。

Q: This provided screenshot is the result of using the pre-trained model with the own collected 16-beam LiDAR data. As can be seen, the MOS results are not good. How to get better results on 16-beam LiDAR data.
A: The experiments show that the generalization of the pre-trained model is not very good for different types of LiDAR scanners. To make it work with 16-beam LiDAR data, one could label several binary labels on own collected data, and fine-tune the pre-trained model.

yuhang9803, 2021-09-12:
> 你好，我用你训练好的模型跑自己的16线雷达的数据，根据label输出对当前帧点云每个点区分静动态，最后输出静动态点云。但是效果是这样的，彩色的是动态点，请问可能是什么原因呢？
> [image]

你好，请问你是怎么把这个模型部署在ros上，然后输出ros格式的topic，然后在rviz上显示出来的呢？可以交流下吗？谢谢了

ypfsmile, 2021-12-10:
[image]
通过将32线数据，转换为需要的输入格式，基于预训练模型测试下来，发现几乎所有点都被认为是动态物体，有可能是模型泛化不好产生么，个人总感觉是数据输入错误。
By converting the vlp-32 data into the required input format and testing it based on the pre-training model, it is found that almost all points are considered dynamic objects. Is it possible that the model generalization is not good? Personally, I always feel that it is a data input error. Have you encountered similar problems?

closed 2022-02-23 by Chen-Xieyuanli