PRBonn/LiDAR-MOS #23: Questions about LIDAR-MOS visualization

MrNeoJeep, 2021-09-19:
Hello, your LIDAR-MOS code is very good, but I have a problem that cannot be visualized when reproducing your code, as shown in the figure:
[image]
After running this command, the program seems to be stuck. I don't know why, I want to get your visualized results, as shown below:
[image]
ps：Author reply：From the results of the operation, it seems to be a tkinter problem.But the tkinker module does not seem to be missing.
If anyone knows how to solve it, hope you can help me, thanks.

Chen-Xieyuanli, 2021-09-19:
Hey @MrNeoJeep,

Thanks for reporting the issue here. 

You could check the version of your tinkter using:
`>>> import tkinter
>>> tkinter.TkVersion`

My tinkter version is 8.6.

MrNeoJeep, 2021-09-20:
Hey@Chen-Xieyuanli
my version is 8.5,tinkter doesn't seem to be abnormal.

Chen-Xieyuanli, 2021-09-20:
Since the visualizer also exploits the graphic card, it could be a problem of that too.

As far as I know, if you run the visualizer remotely (for example via ssh), there will be an OpenGL-related problem.

Could you please try with another computer to see whether the issue is still there?

HSmallYesterday, 2021-12-09:
I meet the same problem, have you solved it?

HSmallYesterday, 2021-12-09:
I got it. I installed vispy through pip, but in fact its dependencies are not completely set up. 
The correct installation setps can be checked in this website https://vispy.org/installation.html.

@Chen-Xieyuanli than you for your work

Chen-Xieyuanli, 2021-12-09:
@HSmallYesterday, Thanks for using our code!

Venkat-S, 2022-02-13:
Hello @Chen-Xieyuanli,

I installed Vispy as @HSmallYesterday suggested. I don't see any error but still i am not able to visualize. 
It would be really helpful if someone could comment on how to resolve the issue.
[image]
.

boazMgm, 2022-02-22:
Hi,
I'm having the exact same issue.
I have installed: 
vispy 0.9.7
tkinter 8.6
is there any solution?
Thanks

Venkat-S, 2022-02-24:
> I have solved the issue by installing pyqt. Thanks

Thank you very much @boazMgm, This solved my issue.

hshac, 2022-03-02:
> Hi, I'm having the exact same issue. I have installed: vispy 0.9.7 tinkter 8.6 is there any solution? Thanks

I install pyqt5 ,but it dose not work.

hshac, 2022-03-02:
It would be really helpful if you can give some suggestions.@boazMgm@HSmallYesterday@Chen-Xieyuanli

LiXiang0021, 2022-07-13:
Did you sloved this problem? @hshac

LiXiang0021, 2022-07-13:
Did you solved this problem? @hshac