PRBonn/LiDAR-MOS #24: Question about loading the pretrained salsanext model

maneekwant, 2021-09-20:
Hi!

Thanks so much for the codes! I've a question about loading the pretrained salsanext model. When I followed the steps outlined in the "How to use" section and tested on the toy example, I came across the issue when trying to run infer.py on the toy dataset (python3 infer.py -d ../../../../data -m ../../../../data/model_salsanext_residual_1 -l ../../../../data/predictions_salsanext_residual_1_new -s valid) and got this error: 

RuntimeError: ../../../../data/model_salsanext_residual_1/SalsaNext_valid_best is a zip archive (did you mean to use torch.jit.load()?)

I tried to switch from torch.load() to torch.jit.load() in the user.py as it suggested but it leads to other errors. What did I do wrong, or did I miss something along the way? I set up the environment according to the instruction linked on the github (using Pytorch 1.1). 

Thank you in advance for your help!

Chen-Xieyuanli, 2021-09-20:
Hey @maneekwant, thanks for using our code!

You need to unzip the pre-trained model before using it.

If there are still any problems, please feel free to let me know.

maneekwant, 2021-09-20:
Thanks for the quick reply! I tried unzipping the SalsaNext_valid_best and now the error is that it’s a folder instead of a file. Did I miss something else too? Thanks so much!

Chen-Xieyuanli, 2021-09-20:
> Thanks for the quick reply! I tried unzipping the SalsaNext_valid_best and now the error is that it’s a folder instead of a file. Did I miss something else too? Thanks so much!

Sorry, I misunderstood your question. You need to only unzip model_salsanext_residual_1 but not SalsaNext_valid_best.

It seems your torch cannot load it and could be a version conflict. Mine is Version: 1.5.1+cu101. You may upgrade the pytorch and give it another try.

maneekwant, 2021-09-24:
That works thanks so much!

Chen-Xieyuanli, 2021-09-24:
Thank you for using our code! If you find the code and my answer helpful, I will appreciate it if you could star our repo :-)

llxClover, 2021-10-22:
> That works thanks so much!

@maneekwant ，I meet the same problem, "FileNotFoundError: [Errno 2] No such file or directory: '../../../../data/model_salsanext_residual_1/SalsaNext_valid_best'",,,,,,,
  could you tell me how to solve?

closed 2021-11-15 by Chen-Xieyuanli