PRBonn/LiDAR-MOS #32: prediction labels in toy dataset

SM1991CODES, 2021-11-07:
Hi,

I see there are some segmentation results already present in the toy dataset.
[image]

The one that says salsanext, does it use one residual image?

Best Regards
Sambit

SM1991CODES, 2021-11-07:
Furthermore, I am trying to convert your motion predictions to BEV space and benchmark it using the eval code present in semantic kitti.
The problem is I am getting very low iou of 29% (0.29). I expect somewhere in the range of 40-50 at least.
Attached is a small video of the predictions against ground truth.
Left hard is gt, right half is predictions from your dataset.

Please help me understand this low score.

https://user-images.githubusercontent.com/69435296/140658495-983676ee-62d1-4f32-b1e2-2e63f68a0670.mp4

Is there a better network? Could you please provide me the labels generated with that network?

Best Regards
Sambit

SM1991CODES, 2021-11-07:
Update - after ignoring gt frames with no motion pixels in it, I get an iou of 53. I think this is okay now.
However, I am still a bit surprised by the many false positives as we can see above in the video.
Best Regards
Sambit

Chen-Xieyuanli, 2021-11-08:
> Hi,
> 
> I see there are some segmentation results already present in the toy dataset. [image]
> 
> The one that says salsanext, does it use one residual image?
> 
> Best Regards Sambit

Hey @SM1991CODES, thanks for using our code. It's interesting that, I didn't find the image_salsanext_preds from the toy dataset. Where did you download the toy dataset?

Chen-Xieyuanli, 2021-11-08:
> Update - after ignoring gt frames with no motion pixels in it, I get an iou of 53. I think this is okay now. However, I am still a bit surprised by the many false positives as we can see above in the video. Best Regards Sambit

Thanks for testing our code on BEV. It's very interesting to see the result of another setup. One possible reason for the false positives could be that we are using the range image representation and the uncertainty of the boundary will cause some shadow-like artifacts.

SM1991CODES, 2021-11-08:
For the dataset, I just downloaded the toy dataset provided in your repo. On extracting, I see the folder structure as shown above.
In the model_salsanext_residual_1, there are your predicted .label files.
I just converted these to BEV and evaluated against GT.

Could you please provide me similar .label predictions for Rangenet++ and MINet? I am trying to evaluate all these models on BEV space.

Best Regards
Sambit

Chen-Xieyuanli, 2021-11-15:
> For the dataset, I just downloaded the toy dataset provided in your repo. On extracting, I see the folder structure as shown above. In the model_salsanext_residual_1, there are your predicted .label [...]

Sorry for the late reply. For the predictions of the Rangenet++, you could directly run the [inferring script](https://github.com/PRBonn/LiDAR-MOS#inferring) with the provided [pretrained rangenet++ model](https://www.ipb.uni-bonn.de/html/projects/LiDAR-MOS/model_rangenet_residual_1.zip) to get the predictions. Since our minet paper is still under reviewing, we currently has no plan to release it.
As also mentioned in the paper, the results of rangenet++ and minet are worse than salsanext.

Since the question is not related to the toy dataset, I would like to close this issue. If you have any further questions, please feel free to ask me to reopen it.

closed 2021-11-15 by Chen-Xieyuanli