PRBonn/LiDAR-MOS #40: multi residual_images, parameters about n_input_scans? thanks!

mengshiyu0109, 2022-02-06:
Dear author ,

I see that it uses 8 residual_imgs in the paper,

So I also want to use 8 images.

But I don't know how to use about the 'n_input_scans' in the code.

Does  "n_input_scans" mean that the multi_residual_image?

[image]

Chen-Xieyuanli, 2022-02-06:
Yes, 'n_input_scans' is to set the number of residual images you want use. You could set it in the config files.

mengshiyu0109, 2022-02-06:
> Yes, 'n_input_scans' is to set the number of residual images you want use. You could set it in the config files.
Thanks for reply!!!!! Happy new year of the tiger.
So I need use "current_index = start_index + self.n_input_scans - 1 "
instead of 
"current_index = start_index"

Similarly，
use "for index in range(start_index, start_index + self.n_input_scans): " 
    not     "for index in range(start_index, start_index + 1)"

Chen-Xieyuanli, 2022-02-06:
Happy New Year!

Yes, we use 'start_index + self.n_input_scans' to avoid loading the beginning of sequences where there are no residual images available.

Since this issue is duplicated with #41, I would close this one, and let's keep track of #41.

closed 2022-02-06 by Chen-Xieyuanli