PRBonn/LiDAR-MOS #41: dims of multi_residuals_images, thanks!

mengshiyu0109, 2022-02-06:
Dear author,

IF n_input_scans =2, 
so  dims of proj_full  is 12? (x,y ,z ,r,e, x,y,z,r,e, residual1 ,resudual2)? 

Is that right?

I'm so sorry to bother you. That's really confused me.
[image]

Thanks.

Chen-Xieyuanli, 2022-02-06:
Not really, we only use the current frame with extra residual images.
If n_input_scans = 2, it will be (x, y, z, r, e, residual1, residual2), which you could find in our paper, section III.C.

mengshiyu0109, 2022-02-06:
> 

Yes ,I think I got the meaning of the Section III. C.

So under the  e Section III. C.  If I use  n_input_scans=2

I just need to change the arcg_cfg.yaml ,    and the code in"class SemanticKitti(Dataset)", like below?
(line 4 and  10  or line 5? ) .
[image]

mengshiyu0109, 2022-02-06:
Dear author, I thought I have misunderstanding the code and code comments in lines 230,2301 234 235 236 &332. ( under the sequences of n_input_scans= 2)
[image]
[image]

in line 332, there is a "concat' operation, so If there is a looping in Line 236.
the dim of  'pro_full' will be added with( x, y,z, r,e,  x,y,z,r,e, ....). ? right ?

mengshiyu0109, 2022-02-06:
> Dear author, I thought I have misunderstanding the code and code comments in lines 230,2301 234 235 236 &332. ( under the sequences of n_input_scans= 2) [image] [image]
>
> in line 332, there is a [...]

So under the sequences of n_input_scans= 2.
under my understanding of the paper Section III.C , I think I will use the line 230 + line 235?  but I got the data flow  about the "proj_full". it's dims is  confused.

Chen-Xieyuanli, 2022-02-06:
In our method, we only use residual images but not full multiple scans. 
We've however also tried to directly concatenate multiple scans together as you mentioned 230+235 and didn't get much improvement.

You could give it a try, and I only test it for two scans for an ablation which could be buggy for more scans.

I hope my answer helps.

mengshiyu0109, 2022-02-06:
> 在我们的方法中，我们只使用残差图像而不是完整的多次扫描。 然而，我们也尝试将多个扫描直接连接在一起，正如您提到的 230+235 并且没有得到太大的改进。
> 
> 你可以试一试，我只对它进行两次扫描以进行消融测试，这对于更多扫描可能是错误的。
> 
> 我希望我的回答有帮助。

closed 2022-02-06 by mengshiyu0109

mengshiyu0109, 2022-02-06:
> In our method, we only use residual images but not full multiple scans. We've however also tried to directly concatenate multiple scans together as you mentioned 230+235 and didn't get much [...]

Dear author,

Maybe I misunderstand the  meanning of the " multi scan and multi residuals images"
I just want to follow the work in the section III.C  with use multi residual images.
[image]

So now  If I just  have the two residual images ,residual_1,residual_2,              just   look at the image,  like the code in line 243.
I dont need to change the Line 231& 236, just  only  change the  n_inupt_scans to 2 in  arch_cfg.yaml,  ?

If it is right ,I think I have understanding the meaning.!!!

thanks for bother you again.

Chen-Xieyuanli, 2022-02-06:
Yes, if you only want to use residual images as what we used in our paper, you need then just change the config file.

可能中文沟通更容易一点，如果你只是想跟我们一样用当前帧的（x, y, z, r, e）加residual images。你只需要修改配置文件里面的n_input_scans就行了。
如果你想用多帧激光的信息（x1, y1, z1, r1, e1, x2, ..., e2）你可以用我们之前测试的代码230 + 235，但是这不是我所提的方法，我只测试过两帧的，对于多帧的情况可能会有问题。

希望我的回答是清楚的，如果你觉得有帮助，请帮我点个赞，谢谢！

mengshiyu0109, 2022-02-06:
> 

谢谢您，终于理解了，finally,
thanks!!!!

noobth1nker, 2024-04-18:
请问一下，对于每帧生成8张残差图像的代码还没有开源是吗？我自己改了一下代码，但是和论文中8张残差图像的效果不太一样。