PRBonn/LiDAR-MOS #42: Generating Residual Images for Inference

boazMgm, 2022-02-21:
Hi,
Say that I'm interesting in using the pretrained model with my own dataset only for inference in the first stage.
Do I have to prepare the residual images beforehand or does the model generates them automatically?
assuming I want to use residual images (residual: True).

Thanks :)

MaxChanger, 2022-02-21:
Yes, you cloud see [utils/gen_residual_images.py](https://github.com/PRBonn/LiDAR-MOS/blob/main/utils/gen_residual_images.py) for more information.

Chen-Xieyuanli, 2022-02-23:
Hi @boazMgm, as @MaxChanger said, you need to generate residual images using our script. You could also integrate it into your online pipeline, but so far we don't have a plan to do that.

closed 2022-02-23 by Chen-Xieyuanli