PRBonn/LiDAR-MOS #43: Understanding the Labels Visualization

boazMgm, 2022-02-23:
Hi,
I'm trying to understand the output of the "visualize_mos.py".
I get the following image:
[image]
1. are the pixels in red in the bottom figure considered to be dynamic?
2. what is the the top figure represents? is it the residual image in the current frame? (I'm working with 1 residual image)

Thanks!

Chen-Xieyuanli, 2022-02-23:
Hey @boazMgm, thanks for using our code.

The top figure shows the range image, the second shows the lidar mos prediction. The red points are predicted as moving.

I hope my answers are clear and helpful.

boazMgm, 2022-02-23:
Hi,
Thanks for your answer!
one more question please.
Do you have a script that takes the label files and gives the scan (x,y,z,e) with another column as static\moving?
Thanks again

Chen-Xieyuanli, 2022-02-23:
Yes, in the data loader, we gather all the information, and you could find it here: https://github.com/PRBonn/LiDAR-MOS/blob/37fae5128ee80a270c22d514ed19fc3cf8ea6dee/mos_SalsaNext/train/tasks/semantic/dataset/kitti/parser.py#L348

Since the original issue is clear, I would like to close it. Feel free to reopen it if needed.

If you find my answer helpful, I will appreciate it a lot if you could give our repo a star :-)

closed 2022-02-23 by Chen-Xieyuanli