PRBonn/LiDAR-MOS #44: Labels Files while Inferring

boazMgm, 2022-02-24:
Hi,
while trying to infer my dataset the code fails at the start with this error:
"File "../..//tasks/semantic/dataset/kitti/parser.py", line 197, in __init__
    assert(len(scan_files) == len(label_files))"

I think there is something basic that I'm missing here.
if the purpose of the inference is to create the label files (predictions), then why it tries to locate them at the beginning?
Thanks

Chen-Xieyuanli, 2022-02-24:
Hey @boazMgm, the label_files here are the ground truth labels for evaluations. I guess you are using the pipeline of training. You may check the inferring pipeline.

boazMgm, 2022-02-24:
Thanks for the reply.
The error occurred while running infer.py file.

Chen-Xieyuanli, 2022-02-24:
> 

Please check which split you are using, if it's the sequences of training, the parser will check the ground truth labels. For testing, there are no ground truth labels thus no checking there.

We stick to the SemanticKitti parser, and it may not be the best setup.

closed 2022-02-27 by boazMgm