PRBonn/LiDAR-MOS #46: Migrating the model to Livox Horizon

Psyclonus2887, 2022-03-15:
Thanks for your excellent work. I download and infer the toy dataset you provide and the performance is good. Now I want to use the model in my Livox Horizon with 80x25 FoV and similar point density to 64-line LiDAR. When I use your tools to generate range image and residual image, I just change the pose.txt and calib.txt to my own and everything is alright, I can generate correct range image and residual image. But when I try to infer my data using the model, I meet this error the range image generation function in salsanext. Are there any possible reasons?
[image]

Chen-Xieyuanli, 2022-03-15:
Hey @Psyclonus2887,

The pretrained model is with a fixed input size of the range image. 
As the error reports, the do_range_projection function does not work for the Livox LiDAR. You need to first change the range projection function for Livox LiDAR.
When using a new sensor, you may also need to train a new model, since the sizes and distributions of point clouds are totally different from that of Velodyne LiDARs.

Psyclonus2887, 2022-03-16:
Thanks a lot. Are there any differences between the range projection method in utils/gen_residual_images.py and that in salsanext? Cause when using utils/gen_residual_images.py, I can generate good range projection images as follows. So I think it may not be the problem of range projection function.
[image]

Chen-Xieyuanli, 2022-03-17:
There should not be a big difference between these two range projection functions and your range projection looks okish.

But you could also see that there are lots of empty pixels and after the spherical projection the range information of the Livox sensor is very sparse.
I would guess LMNet will not work well with this type of input data. I guess use directly the range image from Livox is even better than using our spherical range projection.

Psyclonus2887, 2022-03-17:
Thank you. I have found the reason. The point cloud from Horizon contains some nan points where the depth of these points are 0. In the function in utils, there is a limitation with maxrange and minrange. But in the function in salsanext, there are no such limitation, which leads to illegal input when generating range image. I cut the noise points when generating bin files from Horizon. And the infer procedure can be done.

Psyclonus2887, 2022-03-17:
You are right, the result of prediction is not good.
[image]
But from the visualization result all the point cloud are classified into dynamic, that is weird. Maybe it is because I don't set the configuration right. The point density from Horizon LiDAR is similar to HDL-64, in a accumulation time of 0.1s. And the residual image when N=1 is like follows.
[image]
so what can the problem be? I have several thoughts:
1. The characteristic of non-repeat scanning mode. Although the point density is enough, the distribution makes it different  to infer by your network.
2. The azimuth limitation. the 80 degree FoV reduce the performance of your network.
Any advice will be appreciate. Thank you very much! BTW, what do you mean when you say "use directly the range image from Livox is even better than using our spherical range projection"?

Chen-Xieyuanli, 2022-03-17:
Thanks for the feedback. 

Yes, the pre-trained model was trained on totally different LiDAR data with different distributions. It is expected to perform badly.

Possible solution:
1. using denser and correct range projection of Livox LiDAR and re-train our network. (As shown below, it's from the Livox website and I have no idea how to generate it)
2. using a 3D network that is not bound with the range representation. (We just submit one work to RAL/IROS and will release the code later)

[image]

Psyclonus2887, 2022-03-18:
Thank you very much for your fast reply. I will try both these methods. Looking forward to your new work!

Chen-Xieyuanli, 2022-03-18:
Thanks for using our code. 

Since it's more about another setup, I would like to close this issue.

If you find our code and answers helpful, I'll appreciate it if you could give our repo a star to let more people use and push it forward.

Feel free to reopen this issue if needed.

closed 2022-03-18 by Chen-Xieyuanli