PRBonn/LiDAR-MOS #49: TRAIN BUGS

LiXiang0021, 2022-04-12:
Thanks for your quick response!
@Chen-Xieyuanli

I trained with salsa_mos in semantic kitti. When it run here
Lr: 3.977e-03 | Update: 3.258e-04 mean,5.209e-04 std | Epoch: [0][950/2391] | Time 0.641 (0.623) | Data 0.081 (0.067) | Loss 0.6863 (0.9777) | acc 0.830 (0.855) | IoU 0.417 (0.434) | [1 day, 3:19:41]
LiDAR-MOS/mos_SalsaNext/train/tasks/semantic/../../common/laserscan.py:166: RuntimeWarning: invalid value encountered in true_divide pitch = np.arcsin(scan_z / depth)
i got error
File "conda/envs/salsa/lib/python3.9/site-packages/torch/_utils.py", line 457, in reraise raise exception IndexError: Caught IndexError in DataLoader worker process 1.
and
File "LiDAR-MOS/mos_SalsaNext/train/tasks/semantic/../../common/laserscan.py", line 201, in do_range_projection self.proj_range[proj_y, proj_x] = depth IndexError: index -2147483648 is out of bounds for axis 0 with size 64
The conda env is
`- python=3.9.12=h12debd9_0

python-flatbuffers=1.12=pyhd3eb1b0_0
pytorch=1.11.0=py3.9_cuda11.3_cudnn8.2.0_0
pytorch-mutex=1.0=cuda
tensorflow-base=2.6.0=mkl_py39h3d85931_0
tensorflow-estimator=2.6.0=pyh7b7c402_0`
Is there any wrong with this env？
Or it is too new?
Or there is something wrong with data in residual images?

MaxChanger, 2022-04-12:
The error message shows that the array is out of bounds, but it is difficult to identify the problem with the error message provided. 
Preliminary judgment, this may not involve residual_image.
Maybe the issue occurs in the 985th iteration of epoch 0, indicating that the previous data is loaded and trained normally, can you confirm the sampled/inputed scan (seq_id and scan_id)  in the 950th iteration, and then debug the function do_range_projection separately to get more information.
In addition, can you provide the yaml file used, because proj_x and proj_y are also calculated by fov_up and fov_down.
https://github.com/PRBonn/LiDAR-MOS/blob/16f7c5ddf6d25d30cd05ade39dab7f0a346e7d6c/mos_SalsaNext/train/common/laserscan.py#L169-L174

LiXiang0021, 2022-04-14:
Thanks for you quick response!

The yaml file used is salsanext_mos.yml provided in this mos-SalaNext of LiDAR-MOS project.
And the fov_up and fov_down are 
 
https://github.com/PRBonn/LiDAR-MOS/blob/16f7c5ddf6d25d30cd05ade39dab7f0a346e7d6c/mos_SalsaNext/salsanext_mos.yml#L46-L58
The only difference is i changed the bitch_size to 8.
I going to debug the function of do_range_projection.

MaxChanger, 2022-04-14:
Thanks for sharing this problem, hope to find the problem by debugging do_range_projection.
If you have any progress and questions, welcome to sync with me.

LiXiang0021, 2022-04-22:
After downloading the semantickitti dataset again, I can train this network successfully. So, there is something wrong with the previous dataset.
And, I wonder how to change the input residual frames number of network.
Thanks for your generous helping!

MaxChanger, 2022-04-22:
Hello, glad to hear that you solved the problem. I guess the original data may have been overwritten at some point when processing the data. But it doesn't matter now.

If you want to change the number of input residual images, you could change these two lines:

https://github.com/PRBonn/LiDAR-MOS/blob/16f7c5ddf6d25d30cd05ade39dab7f0a346e7d6c/mos_SalsaNext/salsanext_mos.yml#L25

https://github.com/PRBonn/LiDAR-MOS/blob/16f7c5ddf6d25d30cd05ade39dab7f0a346e7d6c/mos_SalsaNext/salsanext_mos.yml#L73

LiXiang0021, 2022-04-22:
whether I need to change the parameter in 
https://github.com/PRBonn/LiDAR-MOS/blob/16f7c5ddf6d25d30cd05ade39dab7f0a346e7d6c/config/data_preparing.yaml#L14
to the correspond number of residual frames in the network.
I had changed what you said before, but got this error below
 `File "/home/lijianguo/anaconda3/envs/salsa_t/lib/python3.9/site-packages/torch/_utils.py", line 457, in reraise
    raise exception
IndexError: Caught IndexError in DataLoader worker process 0.`
`File "../..//tasks/semantic/dataset/kitti/parser.py", line 244, in __getitem__
    exec("residual_file_" + str(i+1) + " = " + "self.residual_files_" + str(i+1) + "[seq][index]")
  File "<string>", line 1, in <module>
IndexError: list index out of range`

MaxChanger, 2022-04-22:
@LiXiang0021 Hi, sorry I forgot to remind you that these residual images are generated in advance.
If you only generate residual plots for frame t and frame t-1, you will definitely get error messages about not finding the file or list out of bounds when the program runs.

MaxChanger, 2022-04-22:
If you want to use 8 residual images, you could change the `num_last_n` in data_preparing.yaml and re-execute the `gen_residual_images.py`

LiXiang0021, 2022-04-22:
Yes, thanks for your quick respond!

Chen-Xieyuanli, 2022-04-30:
Thanks @MaxChanger @LiXiang0021 for the fruitful discussion. 

Since this issue seems to be solved, I would like to close it. Feel free to ask me to reopen it if there are any further questions.

closed 2022-04-30 by Chen-Xieyuanli