PRBonn/LiDAR-MOS #53: how to change the "n_input_scans"?

sunyuhua-0131, 2022-07-11:
I only change the "arch_cfg.yaml" in model ,but there is a error:size mismatch for module.downCntx.conv1.weight: copying a param with shape torch.Size([32, 6, 1, 1]) from checkpoint, the shape in current model is torch.Size([32, 13, 1, 1]),so how to change the "n_input_scans"?thank you :)

MaxChanger, 2022-07-11:
Hi @pikaqiu0131 , could you provided the cfg.yaml you used.
It should be noted that there are two `n_input_scans` keys in the config, you should keep them consistent, and change them at the same time.
https://github.com/PRBonn/LiDAR-MOS/blob/c638e8b5cecc26b4b11c4421aa37f9c51acddc87/mos_SalsaNext/salsanext_mos.yml#L25

https://github.com/PRBonn/LiDAR-MOS/blob/c638e8b5cecc26b4b11c4421aa37f9c51acddc87/mos_SalsaNext/salsanext_mos.yml#L73

pikaqiu0131, 2022-07-11:
> I change both yaml in my dataset ,but there are more bug than past :

Traceback (most recent call last):
  File "infer.py", line 145, in <module>
    user.infer()
  File "../../tasks/semantic/modules/user.py", line 107, in infer
    to_orig_fn=self.parser.to_original, cnn=cnn, knn=knn)
  File "../../tasks/semantic/modules/user.py", line 134, in infer_subset
    for i, (proj_in, proj_mask, _, _, path_seq, path_name, p_x, p_y, proj_range, unproj_range, _, _, _, _, npoints) in enumerate(loader):
  File "/home/sunyuhua/anaconda3/envs/salsanext/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 345, in __next__
    data = self._next_data()
  File "/home/sunyuhua/anaconda3/envs/salsanext/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 856, in _next_data
    return self._process_data(data)
  File "/home/sunyuhua/anaconda3/envs/salsanext/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 881, in _process_data
    data.reraise()
  File "/home/sunyuhua/anaconda3/envs/salsanext/lib/python3.7/site-packages/torch/_utils.py", line 395, in reraise
    raise self.exc_type(msg)
IndexError: Caught IndexError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/home/sunyuhua/anaconda3/envs/salsanext/lib/python3.7/site-packages/torch/utils/data/_utils/worker.py", line 178, in _worker_loop
    data = fetcher.fetch(index)
  File "/home/sunyuhua/anaconda3/envs/salsanext/lib/python3.7/site-packages/torch/utils/data/_utils/fetch.py", line 44, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/sunyuhua/anaconda3/envs/salsanext/lib/python3.7/site-packages/torch/utils/data/_utils/fetch.py", line 44, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "../..//tasks/semantic/dataset/kitti/parser.py", line 246, in __getitem__
    exec("residual_file_" + str(i+1) + " = " + "self.residual_files_" + str(i+1) + "[seq][index]")
  File "<string>", line 1, in <module>
IndexError: list index out of range

I use this code:python3 infer.py -d ../../../../data -m ../../../../data/model_salsanext_residual_8 -l ../../../../data/predictions_salsanext_residual_8_new -s valid

my "salsanext_mos.yml":

################################################################################
# training parameters
################################################################################
train:
  loss: "xentropy"       # must be either xentropy or iou
  max_epochs: 150
  lr: 0.01               # sgd learning rate
  wup_epochs: 1          # warmup during first XX epochs (can be float)
  momentum: 0.9          # sgd momentum
  lr_decay: 0.99         # learning rate decay per epoch after initial cycle (from min lr)
  w_decay: 0.0001        # weight decay
  batch_size: 24          # batch size
  report_batch: 10       # every x batches, report loss
  report_epoch: 1        # every x epochs, report validation set
  epsilon_w: 0.001       # class weight w = 1 / (content + epsilon_w)
  save_summary: False    # Summary of weight histograms for tensorboard
  save_scans: True       # False doesn't save anything, True saves some
    # sample images (one per batch of the last calculated batch)
  # in log folder
  show_scans: False      # show scans during training
  workers: 4             # number of threads to get data
  
  # for mos
  residual: True # This needs to be the same as in the dataset params below!
  n_input_scans: 8 # This needs to be the same as in the dataset params below!

################################################################################
# postproc parameters
################################################################################
post:
  CRF:
    use: False
    train: True
    params: False # this should be a dict when in use
  KNN:
    use: True # This parameter default is false
    params:
      knn: 5
      search: 5
      sigma: 1.0
      cutoff: 1.0

################################################################################
# classification head parameters
################################################################################
# dataset (to find parser)
dataset:
  labels: "kitti"
  scans: "kitti"
  max_points: 150000 # max of any scan in dataset
  sensor:
    name: "HDL64"
    type: "spherical" # projective
    fov_up: 3
    fov_down: -25
    img_prop:
      width: 2048
      height: 64
    img_means: #range,x,y,z,signal
      - 12.12
      - 10.88
      - 0.23
      - -1.04
      - 0.21
    img_stds: #range,x,y,z,signal
      - 12.32
      - 11.47
      - 6.91
      - 0.86
      - 0.16
      
    # for mos
    n_input_scans: 8 # This needs to be the same as in the backbone params above!
    residual: True   # This needs to be the same as in the backbone params above!
    transform: False # tranform the last n_input_scans - 1 frames before concatenation

thank you very much for your  reply!!!

pikaqiu0131, 2022-07-12:
thank you a lot! I change my document and run it successfully! and If I want to change n=7, I need train my model by myself?or just change the n_inputs in "arch_cfg.yaml" in model 8 ? thank you for your reply:)

MaxChanger, 2022-07-12:
Yeah, I also think you need to re-train the network with `n_inputs_scans=7`, if this repo did not provided this checkpoint in README.
Because the input channel of the Conv2D is changed. If you directly use the pretrained model (n=8) and only change the n-->7, I think there will be an error when loading the pretrained model, you could simplely try and see if it works.

https://github.com/PRBonn/LiDAR-MOS/blob/c638e8b5cecc26b4b11c4421aa37f9c51acddc87/mos_SalsaNext/train/tasks/semantic/modules/SalsaNext.py#L178-L190

closed 2022-08-15 by Chen-Xieyuanli