PRBonn/LiDAR-MOS #60: How to train this code on multi-GPUs？

beyounged, 2022-11-18:
Hi, thank you for your sharing awesome code, I have read this paper and try to train this network from scratch using SalsaNext as the baseline.

However, I find that there is a problem when I train lidar-mos on multi-GPUs. It creates the errors as follow: 
`Ignoring class  0  in IoU evaluation
[IOU EVAL] IGNORE:  tensor([0])
[IOU EVAL] INCLUDE:  tensor([1, 2])
/home/user/.conda/envs/salsanext/lib/python3.7/site-packages/torch/nn/parallel/_functions.py:68: UserWarning: Was asked to gather along dimension 0, but all input tensors were scalars; will instead unsqueeze and return a vector.
  warnings.warn('Was asked to gather along dimension 0, but all '
Traceback (most recent call last):
  File "./train.py", line 178, in <module>
    trainer.train()
  File "../../tasks/semantic/modules/trainer.py", line 274, in train
    show_scans=self.ARCH["train"]["show_scans"])
  File "../../tasks/semantic/modules/trainer.py", line 389, in train_epoch
    loss_m.backward(idx)
  File "/home/user/.conda/envs/salsanext/lib/python3.7/site-packages/torch/_tensor.py", line 255, in backward
    torch.autograd.backward(self, gradient, retain_graph, create_graph, inputs=inputs)
  File "/home/user/.conda/envs/salsanext/lib/python3.7/site-packages/torch/autograd/__init__.py", line 143, in backward
    grad_tensors_ = _make_grads(tensors, grad_tensors_)
  File "/home/user/.conda/envs/salsanext/lib/python3.7/site-packages/torch/autograd/__init__.py", line 37, in _make_grads
    + str(out.shape) + ".")
RuntimeError: Mismatch in shape: grad_output[0] has a shape of torch.Size([2]) and output[0] has a shape of torch.Size([1]).
`
But I train it on single GPU , It can runs and trains.

My environment：
torch                     1.9.0+cu111              pypi_0    pypi
torchaudio                0.9.0                    pypi_0    pypi
torchvision               0.10.0+cu111             pypi_0    pypi

and GPU is RTX3090 (2)

I'd appreciate for you to give me some advice.

Best regards.

MaxChanger, 2022-11-18:
Which config yaml do you use？and how about the batch_size?

beyounged, 2022-11-18:
I use ' salsanext_mos.yml ', and the batch_size = 1.

Thank you for your timely reply!

MaxChanger, 2022-11-18:
bs=1, the data cannot be placed on multiple GPUs, so only one GPU card, or increase batch_size

beyounged, 2022-11-18:
Oh, I get it now and successfully run the lidar-mos, thank you for taking the time to answer this question for me!
I will continue study this wonderful code anyway.

closed 2022-11-18 by beyounged