PRBonn/LiDAR-MOS #61: How to re-train when training is unexpectedly interrupted?

beyounged, 2022-11-19:
Hi，thank you for your wonderful open source，I would like to know whether it can continue to train from the last time when training is unexpectedly interrupted? If can，how to do it ？

beyounged, 2022-11-21:
Hi, I think it may be in logs dir, but I fail to understand very much, could you take little time to help me? Looking forward to your timely reply!

beyounged, 2022-11-21:
Oh, thank you very much, because I am little familiar with pytorch, I will study it in the next time, thank you for your timely help again!

beyounged, 2022-11-21:
Hi~ @MaxChanger , thank you for your guidance, but my true goal is to modify the epoch, the batch_size, and the n_input_scans, and continue to train , so I want to know whether it is possible to do this?

I add "-p" after other parameters on shell, and it create errors again as follows:

`Traceback (most recent call last):
  File "./train.py", line 180, in <module>
    trainer = Trainer(ARCH, DATA, FLAGS.dataset, FLAGS.log, FLAGS.pretrained,FLAGS.uncertainty)
  File "../../tasks/semantic/modules/trainer.py", line 182, in __init__
    self.model.load_state_dict(w_dict['state_dict'], strict=True)
  File "/home/user/.conda/envs/salsanext/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1407, in load_state_dict
    self.__class__.__name__, "\n\t".join(error_msgs)))
RuntimeError: Error(s) in loading state_dict for DataParallelWithCallback:
        size mismatch for module.downCntx.conv1.weight: copying a param with shape torch.Size([32, 10, 1, 1]) from checkpoint, the shape in current model is torch.Size([32, 11, 1, 1]).`

and other little problem is : why[  '--data_cfg', '-dc' is default='config/labels/semantic-kitti-mos.yaml' ] in train.py, I check my File directory only 'config/semantic-kitti-mos.yaml'.

I don't know how to fix these errors, please give me some idea! Looking forward to your reply.

beyounged, 2022-11-22:
Hi~ @MaxChanger , thank you for your guidance, but my expectation is to modify the epoch, the batch_size, and the n_input_scans, and continue to train , so I want to know whether it is possible to do this?

I add "-p" after other parameters on shell, and it create errors again as follows:

Traceback (most recent call last): File "./train.py", line 180, in <module> trainer = Trainer(ARCH, DATA, FLAGS.dataset, FLAGS.log, FLAGS.pretrained,FLAGS.uncertainty) File "../../tasks/semantic/modules/trainer.py", line 182, in __init__ self.model.load_state_dict(w_dict['state_dict'], strict=True) File "/home/user/.conda/envs/salsanext/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1407, in load_state_dict self.__class__.__name__, "\n\t".join(error_msgs))) RuntimeError: Error(s) in loading state_dict for DataParallelWithCallback: size mismatch for module.downCntx.conv1.weight: copying a param with shape torch.Size([32, 10, 1, 1]) from checkpoint, the shape in current model is torch.Size([32, 11, 1, 1]).

I don't know how to fix these errors, please give me some idea! Looking forward to your reply.

MaxChanger, 2022-11-24:
Could you provide more details about "is to modify the epoch, the batch_size, and the n_input_scans, and continue to train"? What I understand is to load the pre-trained model. Do you mean to change these after loading the ckpt model?

MaxChanger, 2022-11-24:
I think it is feasible to change or specify epoch, batch_size should not be affected by loading model, but `n_inputs_scans` cannot be arbitrarily changed, `n_inputs_scans` will change the network structure, thus resulting in different input network model inconsistency. 
https://github.com/PRBonn/LiDAR-MOS/blob/e69d92970422c94685d5d9d13655321a99fff06d/mos_SalsaNext/train/tasks/semantic/modules/trainer.py#L178-L188

beyounged, 2022-11-24:
OK, thank you for your timely reply!

closed 2023-02-13 by Chen-Xieyuanli