PRBonn/LiDAR-MOS #62: How to use the pretrained model to test my own Lidar Scans？

Cxz-dev, 2022-11-26:
Thanks for your work！
                  Now I want to clean my own Lidar Scans.
1.  Use infer.py to get mos predictions label
2. Then use utils/scan_cleaner.py to clean
Is that right? if no, can you give me some advice? Thanks a lot!

Chen-Xieyuanli, 2022-11-27:
Thanks for using our code.

Yes, the mentioned steps are correct. 
One problem you might have is that the pre-trained model may not work well with your own LiDAR data. 
You may fine-tune or train a new model for your case following the training steps.

If you find our code useful, please give it a star. That will help us a lot. Thank you!

Cxz-dev, 2022-11-27:
Oh ,Thanks for your reply. I will have a try!

Cxz-dev, 2022-11-27:
When I use infer.py, I have another problem. 

Traceback (most recent call last):
  File "infer.py", line 144, in <module>
    user = User(ARCH, DATA, FLAGS.dataset, FLAGS.log, FLAGS.model,FLAGS.split,FLAGS.uncertainty,FLAGS.monte_carlo)
  File "/nas_data/chenxz/LiDAR-MOS/mos_SalsaNext/train/tasks/semantic/../../tasks/semantic/modules/user.py", line 40, in __init__
    self.parser = parserModule.Parser(root=self.datadir,
  File "../..//tasks/semantic/dataset/kitti/parser.py", line 462, in __init__
    self.valid_dataset = SemanticKitti(root=self.root,
  File "../..//tasks/semantic/dataset/kitti/parser.py", line 196, in __init__
    assert(len(scan_files) == len(label_files))
AssertionError

I don't know the reason of using ground truth labels provided by SemantiKITTI. If I use my own data, How can I get it?

Cxz-dev, 2022-11-27:
Hi @Chen-Xieyuanli ! When I use infer.py, I have another problem.

Traceback (most recent call last):
File "infer.py", line 144, in
user = User(ARCH, DATA, FLAGS.dataset, FLAGS.log, FLAGS.model,FLAGS.split,FLAGS.uncertainty,FLAGS.monte_carlo)
File "/nas_data/chenxz/LiDAR-MOS/mos_SalsaNext/train/tasks/semantic/../../tasks/semantic/modules/user.py", line 40, in init
self.parser = parserModule.Parser(root=self.datadir,
File "../..//tasks/semantic/dataset/kitti/parser.py", line 462, in init
self.valid_dataset = SemanticKitti(root=self.root,
File "../..//tasks/semantic/dataset/kitti/parser.py", line 196, in init
assert(len(scan_files) == len(label_files))
AssertionError

I don't know the reason of using ground truth labels provided by SemantiKITTI. If I use my own data, How can I get it?

Cxz-dev, 2022-11-27:
> Hi @Chen-Xieyuanli ! When I use infer.py, I have another problem.
>
> Traceback (most recent call last): File "infer.py", line 144, in user = User(ARCH, DATA, FLAGS.dataset, FLAGS.log, [...]

Oh I know the reason. sequence 08 is a valid data. when I change to test, the problem solved.

Chen-Xieyuanli, 2022-11-27:
@Cxz-dev That's great! I'm happy that you fixed it. I will now close this issue but feel free to reopen it if you still get problems here.

closed 2022-11-27 by Chen-Xieyuanli