PRBonn/LiDAR-MOS #63: About set my own Lidar pose to kitti format pose

Cxz-dev, 2022-12-02:
Hi @Chen-Xieyuanli !
        Thanks for your code！
         Now I have another question, I know the code need pose as input(such as gen_residual_images.py), but when I use my own lidar without camera, how can I get the pose.txt from lidar odemetry?
         I think the pose need to be transformed in LiDAR coord, so can I just use the pose(1*3) based in Lidar coord which came from lidar odemetry as input?

Chen-Xieyuanli, 2022-12-21:
Sorry for the late reply. I got stuck by my new career stuff.

Once you get a sequence of LiDAR data, you could use a LiDAR odometry/SLAM method to estimate the 6-DoF poses for each scan. The poses should be in 6 DoF. Otherwise, it would be less accurate when projected from one to another.

Cxz-dev, 2022-12-21:
Thank u for your reply!
         The 6-DoF poses include x, y, z, pitch, roll, yaw. So I should get these, write into a pose.txt. Is that right?

Chen-Xieyuanli, 2022-12-22:
Yes, 6-DoF poses include x, y, z, pitch, roll, and yaw.
More specifically, we use the KITTI format, which uses the 4x4 transformations to describe 6-DoF poses.
In the poses.txt file, for each row, it saves the first 12 elements, the upper 3x4 of the transformation matrix.
The final row of transformation, the homogeneous row, is omitted.

closed 2023-02-13 by Chen-Xieyuanli