PRBonn/LiDAR-MOS #65: how to  visualize?

pikaqiu0131, 2023-02-12:
You should, in operating $ python3 utils/visualize_mos.py -d data -p data/predictions_salsanext_residual_1_valid -s 8 # here we use a specific sequence number, indicate that 

Opening config file config/semantic-kitti-mos.yaml

Sequence folder exists! Using sequence from data/sequences/08/velodyne

Labels folder exists! Using labels from data/predictions_salsanext_residual_8_new/sequences/08/predictions

Using semantics in visualizer

Two navigation:

b: back (previous scan)

n: next (next scan)

q: quit (exit program)

Do you know why

Chen-Xieyuanli, 2023-02-13:
Thanks for using our code. I didn't get your problem. Could you please make it clearer?

pikaqiu0131, 2023-02-13:
sorry！the problem is that i use the "python3 utils/visualize_mos.py -d data -p data/predictions_salsanext_residual_1_valid -s 8 " to visualize ,but the output is:

Opening config file config/semantic-kitti-mos.yaml

Sequence folder exists! Using sequence from data/sequences/08/velodyne

Labels folder exists! Using labels from data/predictions_salsanext_residual_8_new/sequences/08/predictions

Using semantics in visualizer

Two navigation:

b: back (previous scan)

n: next (next scan)

q: quit (exit program)

and there is no image turnout, do you know why?

Chen-Xieyuanli, 2023-02-14:
Are you using a remote server? It seems to be a problem of the X server.

pikaqiu0131, 2023-02-23:
I don't use the remote server...do you know why?