PRBonn/LiDAR-MOS #66: How did you define the moving and static object

Ianpengg, 2023-02-13:
Hello, thank you for your great work.
I'm currently working on a research project related to Moving Object Detection (MOS) using **FMCW Scanning Radar.**
However, I'm facing a challenge in annotating the raw data from the Scanning Radar dataset such as the Oxford Radar Robotcar dataset due to the lack of semantic annotations.
In particular, I'm struggling to differentiate between **moving and static objects**, particularly with regards to small objects like pedestrians.
This is because it can be difficult to distinguish between moving and static objects when the movement between consecutive frames is small.
I would greatly appreciate any tips or techniques for accurately labeling moving and static objects in LiDAR data.
Or can you share what rule do you use to define the moving and static object?
Many Thanks!

Chen-Xieyuanli, 2023-02-13:
Hey @Ianpengg, thanks for your interest in our work!

In the LiDAR-MOS benchmark, we label an object as moving once it moves in the whole recording.

More specifically, we aggregate the point clouds and see whether ghost artifacts exist. If yes, we will label all the corresponding points as moving. Therefore, even if the object moves slowly, we can tell and label it after aggregating all point clouds together.

I hope my answer helps.
Best

Ianpengg, 2023-02-13:
Thank you for your reply !
Here is the sample of my scanning radar data 
I just use 2 scans and aggregate them to check if there are any ghost artifacts exist
where the **light sky blue** represent the overlapped region between current scan and past scan
deep blue represent the current scan's radar image  
and left is my label result
I think this concept is similar to yours? 

[image]

Chen-Xieyuanli, 2023-02-14:
Yes, that's how we labeled the LiDAR data. But it seems too blurry to see the real moving object from the accumulated radar points.

Another way is to record the radar data together with a camera or LiDAR. Recognize the moving objects from other modalities and then label the radar data.

carrolltm, 2023-05-29:
wonderful work, I'm more curious if the car starts to stand still for a while before it starts to move, in this case it is considered that the car is in motion throughout the record, does that mean that this marking method is only suitable for offline, because online cannot see the process of movement behind in time, and the online method will treat the part that is standing still in front as a stationary object?