PRBonn/LiDAR-MOS #67: About using my own lidar data and lidar pose

zyx-99, 2023-02-17:
Thanks for your code!
I want to combine this project with SLAM, I have a question about using my own data. If my laser radar position is calculated by the SLAM system (that is, the position is obtained under the influence of dynamic objects), will it have an impact on this process and cause a large deviation in the segmentation results?
Or if I want to combine this system with SLAM, I need to use the exact position provided by other high-precision sensors instead of the position calculated by SLAM system

Chen-Xieyuanli, 2023-02-19:
Thanks for your interests in our work.

You could optimize both the MOS and odometry iteratively at the same time. For example, you could first get a rough estimation to clean the scan and then estimate the pose, and do this two steps multiple times, similar to our work “Fast Sparse LiDAR Odometry Using Self-Supervised Feature Selection on Intensity Images,”[paper](https://www.ipb.uni-bonn.de/wp-content/papercite-data/pdf/guadagnino2022ral-iros.pdf)

Yes, you could also use poses from other sources as input.

zyx-99, 2023-02-21:
Thanks for your reply! 
I have one last small question. that is whether the rough estimation pose I get from SLAM needs to correspond to Lidar frames one by one or the pose frames are more than Lidar frames. For example, I have 1500 Lidar frames, but my SLAM estimates  the odometery has only 900 pose, In this case,whether it can be cleaned.

zyx-99, 2023-02-21:
> Thanks for your interests in our work.
>
> You could optimize both the MOS and odometry iteratively at the same time. For example, you could first get a rough estimation to clean the scan and then [...]

Thanks for your reply!
I have one last small question. that is whether the rough estimation pose I get from SLAM needs to correspond to Lidar frames one by one or the pose frames are more than Lidar frames. For example, I have 1500 Lidar frames, but my SLAM estimates the odometery has only 900 pose, In this case,whether it can be cleaned.

Chen-Xieyuanli, 2023-02-24:
We never met such a situation. I will leave this issue open and see whether other users will provide any suggestions.

closed 2024-03-20 by zyx-99