PRBonn/LiDAR-MOS #68: Salsanext

ChiyuanFeng, 2023-02-18:


ChiyuanFeng, 2023-02-18:
The link for Salsanext predtrained model doesn't work? Any advice?

Chen-Xieyuanli, 2023-02-19:
It still works for me. You may try it again or use wget, such as:
```wget https://www.ipb.uni-bonn.de/html/projects/LiDAR-MOS/model_salsanext_residual_8.zip```

ChiyuanFeng, 2023-02-21:
Got it now,
Thanks！

11janson, 2023-02-24:
Hi, guys
I using my own lidar data(16 lines） and lidar pose too，but the link for Salsanext predtrained model(residual_8 and residual_1) doesn't work。
`# range image parameters
range_image:
  height: 16
  width: 1024
  fov_up: 15.0
  fov_down: -15.0
  max_range: 50.0
  min_range: 0.5`
Almost all raw points is predicted moving points. 
why? Looking forward to your reply
[image]

Chen-Xieyuanli, 2023-02-24:
Hey @benemer, you may say something about the generalization abilities of point-based and range-image-based methods and advertise your 4D-MOS here ;-)

11janson, 2023-02-25:
> Got it now, Thanks！
> […](#)

Hey, does it work now?

11janson, 2023-02-25:
> 

Hey, Does it work now?

11janson, 2023-02-25:
> 

> 

Hey ,Does it work now? @ChiyuanFeng

ChiyuanFeng, 2023-02-25:
> wget https://www.ipb.uni-bonn.de/html/projects/LiDAR-MOS/model_salsanext_residual_8.zip

Yes.I can download and use this model.

ChiyuanFeng, 2023-02-25:
> I using my own lidar data(16 lines） and lidar pose too，but the link for Salsanext predtrained model(residual_8 and residual_1) doesn't wor

But I have not use it on my own data.

11janson, 2023-02-25:
> > I using my own lidar data(16 lines） and lidar pose too，but the link for Salsanext predtrained model(residual_8 and residual_1) doesn't wor
> 
> But I have not use it on my own data.

Thanks. Do you have a plan using it on your own data?  It doesn't work on my own data, I'm looking for why

benemer, 2023-02-25:
Hey @11janson,

we experienced that range image-based models do not generalize well if the calibration (sensor location, intensity) changes. Since you also use a different sensor with fewer beams, I highly doubt that this will give good results.

As @Chen-Xieyuanli, we developed 4DMOS for moving object segmentation which is not based on projection but infers moving objects from a  sparse voxelized 4D point cloud (x,y,z,t). This method generalizes quite well on other datasets, even with other sensors. So far I tested it on Velodyne HDL-64E, Ouster OS1 64, and Velodyne HDL32E from various datasets using the same model trained on SemanticKITTI.

[Here](https://github.com/PRBonn/4DMOS) is the link to the repository. You can also send me your own data and I can give it a first try.

Best
Benedikt

11janson, 2023-02-27:
> 

> 

> Hey @11janson,
>
> we experienced that range image-based models do not generalize well if the calibration (sensor location, intensity) changes. Since you also use a different sensor with fewer beams, I [...]

Hey benemer, this is my own point clous data with some moving people.

> Hey @11janson,
>
> we experienced that range image-based models do not generalize well if the calibration (sensor location, intensity) changes. Since you also use a different sensor with fewer beams, I [...]

Thank you @benemer , I have sended my own data to your e-mail mersch@igg.uni-bonn.de, is that right?

benemer, 2023-02-27:
Yes, that's right.

I just checked your data! So 4DMOS is also not working correctly. It finds some moving points but misses most of them and predicts false positives on the ground.

The problem is the way the sensor is mounted. Since you use a vertical 16-beam LiDAR and rotate it, you only observe an object within your (narrow) sensor FOV if the sensor itself is at the correct rotation angle. Remember that 4DMOS needs to see the object in multiple subsequent scans. If your sensor is mounted vertically and the FOV is small, you often only partially observe the moving objects only in a few scans.

Is there a specific reason you mounted the sensor vertically? Mounting it horizontally would at least help in segmenting the moving objects.

11janson, 2023-02-27:
Hey benemer,

Thank you for your reply!

This video shows our handheld scanner porduct, may be that is the reason the sensor is mounted vertically, I am not sure.
I do not understand you say only partially observe the moving objects only in a few scans. All the moving objects can be obtained by our handheld scanner.
And do you have any advices working on the our datas ?

 

 
 
 
 

从QQ邮箱发来的超大附件

FJD Trion S1 Intro Video.mp4 (234.37M, 2023年03月29日 16:45 到期)进入下载页面：https://mail.qq.com/cgi-bin/ftnExs_download?k=673036641f3463c2a068287044390b1c4e440306065d58031b0450555b140d0100011b5c540c001e0e545250530e580253055301622539757c7416301050565d166307442b574d415910600d065c561d5b4002645f&t=exs_ftn_download&code=606db993

benemer, 2023-02-27:
Okay, I see. 

> I do not understand you say only partially observe the moving objects only in a few scans. All the moving objects can be obtained by our handheld scanner.

Your handheld scanner uses a LiDAR sensor with 16 beams which should give you around 32k points per 0.1s. Since the sensor itself is rotating by 180 degrees every second, you change the field of view of the sensor and therefore can measure a larger portion of your environment within one second.

What I mean by partially observing is that you might see a moving object in one scan but not in the next one. This is because your sensor is rotating and the moving objects get out of your field of view (which is anyway small using a 16-beam sensor). This makes it hard for 4DMOS to identify moving objects in a sequence of scans since we assume to see the moving object for multiple, subsequent scans.

benemer, 2023-02-27:
> A LiDAR sensor with 64 beams maybe better, right?
This should help, especially for having more measurements on moving objects which are not close to the sensor, since point clouds from a 16-beam LiDAR become quite sparse in the distance. However, the main problem I see is the rotation of the sensor on the handheld itself.

> And nother question, how about merging subsequent 4(or may be 5) scans in my datas as a scan in your datas? so that I can see a moving object in current 4 scans and in the next 4 scans.
This is hard to say. In 4DMOS we assume that all points in a scan are measured at the same point in time. Strictly speaking, this does not hold for a rotating LiDAR sensor, but the framerate of 10 Hz is quite large and we also compensate for the ego-motion within a single scan revolution. In your case, you would aggregate points from 1s into one virtual timestamp which is in my view not negligible anymore. Also, if there is a moving object in subsequent frames, you would lose this information by merging.

11janson, 2023-02-28:
> This should help, especially for having more measurements on moving objects which are not close to the sensor, since point clouds from a 16-beam LiDAR become quite sparse in the distance. However, [...]

Thank you Benemer. I have already understood what you said.  A moving object should be appear in some subsequent frames at very short intevals.  So the 4DMOS is not suitable for our own datas., right?
Thanks again.

benemer, 2023-02-28:
At least using 4DMOS out of the box does not work. Do you need online MOS? I think in your case it would be easier to loot at some offline map cleaning approaches.

11janson, 2023-03-01:
> At least using 4DMOS out of the box does not work. Do you need online MOS? I think in your case it would be easier to loot at some offline map cleaning approaches.

Yeah. I am looking at some offline map cleaning approaches such as "removert", and have got some results. 
And I am trying another online approaches called "RF-LIO",  maybe it works.

benemer, 2023-03-01:
We are currently working on another MOS approach and might release it in a couple of months. We can also try this on your data. I will let you know as soon as it's available!

closed 2023-03-01 by benemer