PRBonn/LiDAR-MOS #72: About training time

Hellooooooooo-web, 2023-05-23:
Dear author. I want to know how long does it take for you to training data using salsanext?
I am trying to train the data myself. I don't know if it's due to my environment that the speed is so slow.
I havea 3090 GPU, CUDA11.0 pytorch1.7 TensorFlow2.4.0 in batch_size=10. the program shows that the remaining time is approximately 5 days and 12 hours.

MaxChanger, 2023-05-23:
You can refer to this to speed up training, but the performance may be slightly worse than using full data.
https://github.com/haomo-ai/MotionSeg3D/blob/ce43ed4f4f8b693c01854be046b00eaf6ed88cf9/common/dataset/kitti/parser.py#L215-L217

Hellooooooooo-web, 2023-05-24:
> You can refer to this to speed up training, but the performance may be slightly worse than using full data. https://github.com/haomo-ai/MotionSeg3D/blob/ce43ed4f4f8b693c01854be046b00eaf6ed88cf9/common/dataset/kitti/parser.py#L215-L217

Thanks

closed 2023-05-24 by Hellooooooooo-web